STMicroelectronics/cmsis-device-u0 #1: Missing definitions around RCC_CFGR_SWS_HSISYS and RCC_CFGR_SW_HSISYS

UweBonnes, 2024-10-18:
Hello,

u0 headers miss the definitions around RCC_CFGR_SWS_HSISYS and RCC_CFGR_SW_HSISYS, while most other families have these definitions. This needs extra specail handling. Please add.

TOUNSTM, 2024-10-21:
Hello @UweBonne,

 Thank you for your report. 

After reviewing our reference manual and drivers, we found that our code aligns with the current specifications. We could not locate the definitions **RCC_CFGR_SWS_HSISYS**  and **RCC_CFGR_SW_HSISYS** in any of the families we support. 

To assist you better, could you please provide more details or context about the specific family or use case you are referring to? This additional information will help us understand your request better and provide a more accurate solution.

Best regards,

TOUNSTM, 2024-10-21:
See https://github.com/STMicroelectronics/cmsis-device-g0/issues/4

TOUNSTM, 2024-10-21:
Hello @UweBonne,

Thank you for your report.

After reviewing our reference manual and drivers, we found that our code aligns with the current specifications. 

To assist you better, could you please provide more details or context about the specific family or use case you are referring to? This additional information will help us understand your request better and provide a more accurate solution.

Best regards,

UweBonnes, 2024-10-29:
Sorry, my posting was no exact. I miss the definitions
#define RCC_CFGR_SW_MSI                      (0x00000000UL)                    /*!< MSI oscillator selection as system clock */
#define RCC_CFGR_SW_HSI                      (0x00000001UL)                    /*!< HSI16 oscillator selection as system clock */
#define RCC_CFGR_SW_HSE                      (0x00000002UL)                    /*!< HSE oscillator selection as system clock */
#define RCC_CFGR_SW_PLL                      (0x00000003UL)                    /*!< PLL selection as system clock */

#define RCC_CFGR_SWS_MSI                     (0x00000000UL)                    /*!< MSI oscillator used as system clock */
#define RCC_CFGR_SWS_HSI                     (0x00000004UL)                    /*!< HSI16 oscillator used as system clock */
#define RCC_CFGR_SWS_HSE                     (0x00000008UL)                    /*!< HSE oscillator used as system clock */
#define RCC_CFGR_SWS_PLL                     (0x0000000CUL)                    /*!< PLL used as system clock */

that all other STM32 family beside Uo headers have. This requires extra handling for the U0 family. G0 headers for no obvious reason change RCC_CFGR_SWx_HSI to RCC_CFGR_SWS_HSISYS and RCC_CFGR_SWx_PLL to  RCC_CFGR_SWS_PLLRCLK introducing another exception.

TOUNSTM, 2024-11-26:
Hello @UweBonnes ,

These defines are used in legacy series but are no longer valid in newer series like U0 and H7RS. The recommendation is to avoid using defines with fixed values. In CMSIS, it is always necessary to use bit positions and bit masks. This kind of define can be made in the driver but not in CMSIS. Alternatively, you can use the following macros, which are defined in the driver stm32u0xx_hal_rcc.h:

```
/** @defgroup RCC_System_Clock_Source System Clock Source
  * @{
  */
#define RCC_SYSCLKSOURCE_MSI            (0x00000000U)                     /*!< MSI selection as system clock */
#define RCC_SYSCLKSOURCE_HSI            RCC_CFGR_SW_0                     /*!< HSI selection as system clock */
#define RCC_SYSCLKSOURCE_HSE            RCC_CFGR_SW_1                     /*!< HSE selection as system clock */
#define RCC_SYSCLKSOURCE_PLLCLK         (RCC_CFGR_SW_1 | RCC_CFGR_SW_0)   /*!< PLL selection as system clock */
#define RCC_SYSCLKSOURCE_LSI            RCC_CFGR_SW_2                     /*!< LSI selection as system clock */
#define RCC_SYSCLKSOURCE_LSE            (RCC_CFGR_SW_2  |RCC_CFGR_SW_0)   /*!< LSE selection as system clock */
/**
  * @}
  */

/** @defgroup RCC_System_Clock_Source_Status System Clock Source Status
  * @{
  */
#define RCC_SYSCLKSOURCE_STATUS_MSI      (0x00000000U)                      /*!< MSI used as system clock */
#define RCC_SYSCLKSOURCE_STATUS_HSI      RCC_CFGR_SWS_0                     /*!< HSI used as system clock */
#define RCC_SYSCLKSOURCE_STATUS_HSE      RCC_CFGR_SWS_1                     /*!< HSE used as system clock */
#define RCC_SYSCLKSOURCE_STATUS_PLLCLK   (RCC_CFGR_SWS_1 | RCC_CFGR_SWS_0)  /*!< PLL used as system clock */
#define RCC_SYSCLKSOURCE_STATUS_LSI      RCC_CFGR_SWS_2                     /*!< LSI used as system clock */
#define RCC_SYSCLKSOURCE_STATUS_LSE      (RCC_CFGR_SWS_2  |RCC_CFGR_SWS_0)  /*!< LSE used as system clock */
/**
  * @}
  */
```
As there is no plan to add these defines, please allow me to close this issue.

Thank you again for your contribution. We are looking forward to reading from you again.
Best Regards,

closed 2024-11-26 by TOUNSTM