STMicroelectronics/cmsis-device-u0 #2: Windows 1252 encoding in Include/stm32u0xx.h

blackbird998, 2024-11-28:
Hi,
In file `@brief` the ( ’ ) character in `peripheral’s` (lines 11 and 12) is Windows 1252 encoded instead of UTF-8.
```markdown
  *              - To use or not the peripheral’s drivers in application code(i.e.
  *                code will be based on direct access to peripheral’s registers
  *                rather than drivers API), this option is controlled by
  *                "#define USE_HAL_DRIVER"
 ```

Kind regards,
Adam

KRASTM, 2024-12-02:
Hello @blackbird998,

Thank you for the report.

We confirm the point, and it will be fixed.

With regards,

KRASTM, 2024-12-03:
Fixed in commit: https://github.com/STMicroelectronics/cmsis-device-u0/commit/1d5ad465f74a534033c4cbb4aee084c2b6bba3d7

closed 2024-12-03 by KRASTM