STMicroelectronics/cmsis-device-u0 #4: Issue with IRQn_Type DMA

blackbird998, 2024-12-17:
Hi,
In `stm32u073xx.h` and `stm32u083xx.h` in `IRQn_Type` enum interrupt number 11 is defined as `DMA1_Ch4_7_DMA2_Ch1_5_DMAMUX_OVR_IRQn`, and in `stm32u031xx.h` as `DMA1_Ch4_7_DMAMUX_OVR_IRQn`. When using CubeMX to generate `Core/Src/dma.c` file on STM32U031xx it will use `DMA1_Ch4_7_DMA2_Ch1_5_DMAMUX_OVR_IRQn` in `HAL_NVIC_SetPriority()` and `HAL_NVIC_EnableIRQ()`, which is undeclared on this device.

[image]

Kind regards,
Adam

ASEHSTM, 2024-12-20:
Hello @blackbird998, 

Thank you for bringing this to our attention. This issue has been tracked internally.
ST Internal Reference: 199154

With Regards,

ASEHSTM, 2025-01-06:
Hello @blackbird998,

This issue will be fixed in STM32CubeMX V6.14.0. Please allow me to close it.

With Regards,

closed 2025-01-06 by ASEHSTM