STMicroelectronics/cmsis-device-u0 #7: FLASH_ECCR_ADDR_ECC_Msk seems to be too narrow

jamesl-dm, 2025-02-19:
The FLASH->ECCR register has a field ADDR_ECC that gives the location of the first ECC error since the ECCC or ECCD flags were set (for correctable or double error). In the text of the user manual in section 3.3, it says:

> When an ECC error is detected, the address of the failing double word is saved in ADDR_ECC[16:0] bitfield of the FLASH_ECCR register. ADDR_ECC[2:0] are always cleared.

In other words it says ADDR_ECC is 17 bits wide, and the error address 0x08000000 + ADDR_ECC.
However, the table in section 3.7.6, and the header file at line 3662, show the field as 14 bits wide:

https://github.com/STMicroelectronics/cmsis-device-u0/blob/a973f2d9218cad5d05ae7ae9a3c49ca25fa336f1/Include/stm32u083xx.h#L3662

And the largest STM32U0 parts have 256 kB of flash. So they would need **18** bits.

The user manual also contains this confusing line in section 3.7.6 - which I assume is leftover from a different device since there is no mention of the STM32U0 being dual banked:

> This register applies to single-bank products and to Bank 1 of dual-bank products.

So I'm not sure what this means for users of 256 kB chips. It's not clear if the register is actually 18 bits, 17 bits, or 14 bits.

jamesl-dm, 2025-02-24:
Experimentally I'm seeing (FLASH->ECCR & 0xFFFF) == address_of_fault / 8.
So I think the field is actually 15 bits.
For a fault at 0x08020008, I read back 0x4001.

TOUNSTM, 2025-02-25:
Hello @jamesl-dm,

Thank you for your report. We have forwarded your point to our development teams for further investigation. I will keep you informed.

With regards,

TOUNSTM, 2025-04-10:
ST Internal Reference: 207375